matthewlloyd/Llama-Mini-Firmware #1: FR: Stealthchop mode for Extruder for Quieter Operation

talldonkey, 2021-06-29:
FR is for overall end goal for a quieter printer 

Hypothesis:  Switching Extruder to Stealthchop will make it quieter, and hopefully doesn't have downsides.

The stock prusa mini firmware has the Extruder in spreadcycle mode.  I believe the X Y Z axis are already in stealthchop. 
There may be an theoretical opportunity to quiet the extruder down by switching the extruder to stealthchop.  This may require tweaks such as upping the extruder current a bit for reliable operation, if it doesn't work at stock current.

matthewlloyd, 2021-07-07:
I won't have a chance to implement this personally but I'm happy to review any related pull requests. It would need to be made optional and adjustable via the settings menu, and tested to make sure the extruder doesn't skip even more steps than it already does.

espr14, 2022-12-20:
`M569 S1 E` command should do the work.

matthewlloyd, 2023-02-04:
Thanks for opening an issue. Unfortunately since I am no longer have time to maintain this project, I am closing all PRs and issues so that I can archive the repository.

closed 2023-02-04 by matthewlloyd