matthewlloyd/Llama-Mini-Firmware #4: [BUG] Print Time Display

Aquaerics, 2021-07-01:
I have testet the v1.0.1 beta on my prusa mini. 

There is no remain print time in the status display. Only the % will be increased. I use the prusa slicer.

dcs3473, 2021-07-01:
I am getting the same results from Cura.

matthewlloyd, 2021-07-02:
Are you using OctoPrint? Have you read https://github.com/matthewlloyd/Llama-Mini-Firmware#print-progress?

matthewlloyd, 2021-07-07:
Closing since no reply and this is working for me and other users when using the suggested OctoPrint plugins and settings as described in the link above.

Aquaerics, 2021-07-07:
I don't use octoprint. I use the prusa slicer and print from the USB stick. Normaly the prusa displays print time and remain print time on the display while printing. So you can see how long the printer is printing and how long it will take. 
The remain print time should be extracted from the g-code from the prusa slicer.

matthewlloyd, 2021-07-07:
I see, so this is on the regular print status screen rather than the OctoPrint one.

Can you upload a small example G-code file that has the issue?

Aquaerics, 2021-07-11:
Here is a GCODE File sliced for Prusa mini with PETG. Print time and Remain Time will not be display while printing.
It was sliced with prusa slicer 2.3.2

[Prusa_0.2mm_PETG_MINI_24m.zip](https://github.com/matthewlloyd/Llama-Mini-Firmware/files/6796629/Prusa_0.2mm_PETG_MINI_24m.zip)

matthewlloyd, 2021-07-11:
Thanks. Do you see the same issue with the latest version of the stock Prusa firmware? At least one other user is reporting that they are: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1499

matthewlloyd, 2021-07-12:
I was able to reproduce this, both in Llama and in Prusa's stock firmware 4.3.2-RC1, so this is an upstream issue. Fixed in v1.0.5.

closed 2021-07-12 by matthewlloyd