matthewlloyd/Llama-Mini-Firmware #5: [BUG] Bootloop

felixhesshaw, 2021-07-07:
Hello,

I can't get the custom firmware to work with my prusa mini:
After building the firmware, I flashed the bbf file in bootloader mode. This worked normally but then I ended up in a bootloop shortly after that.

Tried your bbf file too, but didn't succeed to boot the prusa normally...

Have anyone else this problem?

Best regards and thanks for your work
Felix

matthewlloyd, 2021-07-07:
I haven't had that issue. What command did you use to build the BBF? At what point does it reset during the bootloop (perhaps upload a video on YT or something)? Are you running an Original Mini or a clone, and what version is the bootloader?

You can always build and flash in DFU mode, which will update the bootloader to the latest version (v1.1). There are instructions here: https://github.com/matthewlloyd/Llama-Mini-Firmware#flashing-in-dfu-mode

felixhesshaw, 2021-07-09:
Hey Matthew,

thanks for your quick reply!
I used this command that I know from the original firmware:
python utils/build.py --final --generate-bbf

Here is a video from the flashing process and the bootloop:
https://streamable.com/ysm7o6

My Prusa Mini is a chinese Clone but I have successfully flashed the original firmware with a few modifications in Configuration_A3ides_2209_MINI.h (like bondtech extruder, PID and thermistor settings).  

My bootloader is 1.0.0. I will try to flash in DFU Mode. 
Is it possible to revert to original firmware then? 

Thanks for your help!

matthewlloyd, 2021-07-09:
I would guess one of these is the issue:

- Bootloader offset. The command you use builds three variants, mini_release_noboot, mini_release_emptyboot, and mini_release_boot. You didn't specify which one you flashed, but only one of these will work (mini_release_emptyboot I think). This is because the code uses absolute addressing, and the start address of the firmware changes depending on the variant. The bootloader always flashes the BBF at a fixed address, so if the code is compiled assuming it starts at a different address, it won't work. This is the most likely explanation.
- Something related to EEPROM size. The Chinese clone may use a smaller EEPROM than the original, so the EEPROM settings might not work correctly. This is unlikely.

It's always possible to revert to the original firmware. Flashing in DFU is the same as flashing with the bootloader, just doing it at a lower level and bypassing the bootloader entirely. The file format is also different (standard DFU, vs Prusa's proprietary BBF). It's the only way to upgrade the bootloader as far as I know (Prusa hasn't released the source code for the bootloader, so who knows).

felixhesshaw, 2021-07-09:
Hey,

I flashed the bbf from emptyboot and tried your release bbf too. I hope it's not the EEPROM ;) That would be really annoying.

Next thing I want to try is building the firmware in debug mode and hope to get some infos about the error.

Do you think it is the outdated bootloader that causes this problem?

matthewlloyd, 2021-07-09:
You could also try cloning the original Prusa firmware repo and building the BBF from that using the exact same command. If that doesn't work either, then it probably anything related to Llama, i.e. it isn't the extra EEPROM space. If this doesn't work, but it does work if you flash with the official BBF, then there's something up with the way you're building it.

Unfortunately without connecting an ST-Link to the SWD debug pins it will be hard to figure out. The error occurs too early to get anything useful out of the USB or serial ports.

Outdated bootloader could cause the issue but seems unlikely. I would probably upgrade by DFU anyway to be sure.

closed 2021-07-10 by matthewlloyd

qp0n4ik, 2021-07-10:
Hey do you have any luck flashing it? I have the same problem, but I have no idea how to flash it in DFU, could you please help me?

felixhesshaw, 2021-07-10:
Didn't tried it with dfu. I decided to stay with my own modified firmware based on the original from prusa. Had no trouble flashing the bbf with that.
Lost a little bit hope with this custom rom despite the features sound really appealing with skew correction and hot end rpm control.

qp0n4ik, 2021-07-10:
> Didn't tried it with dfu. I decided to stay with my own modified firmware based on the original from prusa. Had no trouble flashing the bbf with that.
> Lost a little bit hope with this custom rom despite the features sound really appealing with skew correction and hot end rpm control.

How did you fix the bootloop?

qp0n4ik, 2021-07-10:
Could help me please? Im in constant bootloop and I cant do anything, i could tip you in paypall for your time

felixhesshaw, 2021-07-10:
I just flashed the "old" bbf from my modified prusa firmware.
You can use the original bbf file as well:

Just put it on an usb drive and insert it into your prusa mini.
Then turn on the printer and immediatly push the wheel button several times.
Then you confirm the flashing and you are done.

Feel free to ask again if you're still stuck.

felixhesshaw, 2021-07-10:
Here is a guide line for flashing the firmware (Step 8):
https://help.prusa3d.com/en/guide/how-to-update-firmware-mini-mini_128421

qp0n4ik, 2021-07-10:
oh wow!!!!!  I didn't know about  that "immediately push the wheel button several times" 
thank you for saving me money I thought I have to buy new board now. 

May I ask you to try your custom firmware?  the only reason I wanted this on is to change e-steps because I just installed new bondtech extruder

matthewlloyd, 2021-07-11:
I just released v1.0.4 which fixes a small bug I just noticed, which could in theory cause a bootloop on some devices. I don't know whether this actually is the cause of your bootloop, but if you like you can give it a try once you're comfortable switching between firmwares.

felixhesshaw, 2021-07-11:
Hey Matthew,
I flashed the new version successfully! Big thanks to you for this fix.

@qp0n4ik Do you want to give the fixed version a try? I can compile a modified version of the original prusa for you. But I don't think this thread is the right place for this. Maybe try the fixed version first ;)

matthewlloyd, 2021-07-11:
> I flashed the new version successfully! Big thanks to you for this fix.

Very glad to hear it, and good to know this was the issue after all - sorry for the bug and thanks for your patience.