matthewlloyd/Llama-Mini-Firmware #6: [BUG]<Bootloop plz help>

qp0n4ik, 2021-07-10:
I did everything that instruction says, and now I have constant bootloop, i have no idea how to reflash it in the DFU mode? could anyone make video of the process? do I need raspberry pi for that? I have original prusa mini

matthewlloyd, 2021-07-10:
Did you put the jumper back to its original position? That seems like the most likely explanation.

Unfortunately I have limited time to offer to support the hardware of Prusa Buddy boards and bootloader issues - I recommend contacting Prusa support directly for assistance with the appendix, jumpers, and flashing.

closed 2021-07-10 by matthewlloyd

qp0n4ik, 2021-07-10:
Why you closed the topic? No, I did not put a jumper on original place.

matthewlloyd, 2021-07-11:
I just released v1.0.4 which fixes a small bug I just noticed, which could in theory cause a bootloop on some devices. I don't know whether this actually is the cause of your bootloop, but if you like you can give it a try once you're comfortable switching between firmwares.