matthewlloyd/Llama-Mini-Firmware #8: [BUG]<Enter comprehensive title>

Greeneyedblackcat, 2021-08-07:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version**-  v1.0.5 (based on Prusa buddy 4.3.2-RC1)

**Original or Custom firmware** [Original/Custom]

**Optional upgrades** - Runout sensor

**USB drive or USB/Octoprint**
  Please let us know if you print via USB flash drive or USB/Octoprint

**Describe the bug**
I am having a print fan not running  error come up within the first 4% of the print about five times, my print fan is running, after I clear it five times it will run the rest of the print without a single error. When I switch back to factory prusa firmware the problem goes away. My mini is brand new, I checked the plug, connections and I Ohmed out the wires with a multi meter and they are verified good. I love your Llama Firmware and wish to continue to use it.   Any assistance you could provide me on this matter would be greatly appreciated.       Regards JJK 

**How to reproduce, Simply start a print . Like clockwork Print fan errors,
**Expected behavior**
  A clear and concise description of what you expected to happen.

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.

**Crash dump file**
  Please attach the crash dump file. This will make it easier for us to investigate the bug.

**Video**
  Please attach a video. It usually helps to solve the problem.

matthewlloyd, 2021-08-07:
I'm pretty sure this is a bug in 4.3.2-RC1 on Prusa's side because I haven't made any changes that would affect the print fan code. What specific version of factory Prusa firmware are you switching back to? You can check in the Settings menu or in the bootloader screen.

Greeneyedblackcat, 2021-08-07:
Matthew, Thank you so much for the Quick reply. I'm pretty sure its on Prusas end.  The Prusa firmware is 4.3.1,  I really want to keep your Llama firmware, do you think I could simply download one of your previous versions of Llama to fix this problem,  I may try one of your previous versions, you have done a spectacular job with this Firmware. Don't get me wrong I'm not bitching I am grateful to have it bugs and all.  Thanks again for the Reply.

matthewlloyd, 2021-08-07:
All the Llama versions so far are based on 4.3.2-RC1, so they will all have this issue. The best hope would be to reproduce the bug with Prusa's own 4.3.2-RC1, open an issue on their GitHub, and wait for them to fix it as they prepare to release 4.3.2 final. Once Prusa releases 4.3.2 final, I'll release a new version of Llama.

Greeneyedblackcat, 2021-08-07:
Sounds great, I can just deal with the bug until 4,3,2 final. Thank you so much for your time and effort on these Mini's.  I am a  barely computer literate Safecracker by trade and I Thank god there are computer Geniuses like yourself on this planet, Looking forward to 4.3.2 final in Llama.

matthewlloyd, 2021-08-14:
I released v1.0.6, based on 4.3.2-final from Prusa. Let me know if it fixes this issue for you.

Greeneyedblackcat, 2021-08-14:
Thank you so much. I will install it and let you know. Thanks again for
your help.    Regards.  James Keenan

matthewlloyd, 2021-09-16:
This is over a month old, I'm going to close. If it's not fixed, please let me know.

closed 2021-09-16 by matthewlloyd

Greeneyedblackcat, 2021-09-16:
Matthew its fixed. Thank you so much.  Regards.  JK