matthewlloyd/Llama-Mini-Firmware #11: [BUG]<Mesh bed leveling  hangs the system>

Sorav1, 2021-08-16:
Sir, I have installed the latest version of your firmware but when I do mesh calibration from lcd the system hangs and I need to reset all the time. The issue is not there with original Prusa firmware.

Sorav1, 2021-08-16:
https://youtu.be/JloiyZ9ovZA

matthewlloyd, 2021-08-17:
Do you have skew coefficients enabled? That could potentially cause an out-of-bounds move. Just a hypothesis.

Please connect OctoPrint or some other G-Code terminal and see if there is any error output.

Sorav1, 2021-08-20:
Hi greetings of the day.
I have tried keeping skew coefficients disabled/enabled but still the same issue.

Sorav1, 2021-08-20:
Will connect the machine to pronteface and update you the results.

matthewlloyd, 2021-09-16:
This is a month old, I'm going to close. If it's not fixed, please let me know.

closed 2021-09-16 by matthewlloyd