matthewlloyd/Llama-Mini-Firmware #15: [FEATURE REQUEST] Higher density mesh bed leveling

bbbenji, 2021-12-20:
Please add the option to configure the number of MBL measurement points. 

I see it has been requested and implemented in the MK3 firmware here: https://github.com/prusa3d/Prusa-Firmware/issues/1239
as well as requested for the Mini in the original FW repo here: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1492

In addition to the above, it would be cool to set a 3x3 mesh that measures at the exact location of each heatbed mounting point.

If such a feature was to be implemented, I would gladly update the G29 absolute values to relative converter: https://bbbenji.github.io/PMSBLM/

espr14, 2022-12-21:
The main issue is the location of magnets which interfere with the PINDA. If you want different count of points you have to find proper places. Otherwise, the measurement will be worse.

Eldenroot, 2022-12-22:
@espr14 I would like to see at least 20 or 24 mesh leveling

espr14, 2022-12-23:
25 (5x5) points would make sense for squared bed.

matthewlloyd, 2023-02-04:
Thanks for opening an issue. Unfortunately since I am no longer have time to maintain this project, I am closing all PRs and issues so that I can archive the repository.

closed 2023-02-04 by matthewlloyd