matthewlloyd/Llama-Mini-Firmware #18: [FEATURE REQUEST] ESP Wifi Module

Fra085, 2022-01-04:
Printer type: MINI
Optional upgrades: ESP Wifi Module

It would be great if we could use the intended ESP module to enable wifi printing. 
I don't know what are the complications with it (PRUSA team has not released it yet) but if only it could work through wifi as it works with LAN would be perfect.

espr14, 2022-12-21:
This took a year of development because they had to create completely new FW for the ESP module. Fionally, it's in FW 4.4.0 which could get here soon.

espr14, 2023-01-11:
Here it is:
https://github.com/Llama-FW/Llama-P32-FW/releases/tag/v4.4.01

matthewlloyd, 2023-02-04:
Thanks for opening an issue. Unfortunately since I am no longer have time to maintain this project, I am closing all PRs and issues so that I can archive the repository.

closed 2023-02-04 by matthewlloyd