matthewlloyd/Llama-Mini-Firmware #19: [FEATURE REQUEST] Update with Buddy Firmware 4.3.4

dimitribellini, 2022-01-21:
Do you think we can have your modded firmware based on Prusa Buddy 4.3.4?
Thanks

espr14, 2022-12-21:
Ready to pull #30.

espr14, 2022-12-31:
Here's 4.3.4 with extras:
- fixed skew
- Llama settings available during printing
- Nozzle cooling by fan (`M109 R170 C`)
- game Snake

[Llama_MINI_4.3.4_extra.zip](https://github.com/matthewlloyd/Llama-Mini-Firmware/files/10327981/Llama_MINI_4.3.4_extra.zip)

prongato, 2023-01-01:
> Here's 4.3.4 with extras:
> 
> * fixed skew
> * Llama settings available during printing
> * Nozzle cooling by fan (`M109 R170 C`)
> * game Snake
> 
> [Llama_MINI_4.3.4_extra.zip](https://github.com/matthewlloyd/Llama-Mini-Firmware/files/10327981/Llama_MINI_4.3.4_extra.zip)

Thank you for this!  It’s awesome to have an updated version of Llama!

espr14, 2023-01-04:
@matthewlloyd Will you release this version or do you agree with moving this project to a shared repository? I don't want to just fork the development. This would be confusing for the users (where is the latest version, where to put issues etc.). What do you think?

matthewlloyd, 2023-01-04:
> @matthewlloyd Will you release this version or do you agree with moving this project to a shared repository? I don't want to just fork the development. This would be confusing for the users (where is the latest version, where to put issues etc.). What do you think?

Thanks for your efforts. Given time constraints on my side I think the only course of action open at this time is for you to start a fork. I am happy to add a link to your fork to the README of this repo.

Please note that since this repo is under GPLv3 and you have shared a derived binary above `Llama_MINI_4.3.4_extra.zip`, you need to provide the source code used to build it. I know you have opened PRs for some of these features, but you need to provide a link to a branch or a source archive that contains the complete source code for your binary distribution, including all extras you have added (including the Snake game).

espr14, 2023-01-06:
Now it's forked and released:

https://github.com/Llama-FW/Llama-P32-FW/releases/tag/v4.3.4a

matthewlloyd, 2023-02-04:
Thanks for opening an issue. Unfortunately since I am no longer have time to maintain this project, I am closing all PRs and issues so that I can archive the repository.

closed 2023-02-04 by matthewlloyd