matthewlloyd/Llama-Mini-Firmware #20: Skew compensation doesn't work

dagov42, 2022-01-22:
**Printer type** - MINI+ with SuperPinda

**Printer firmware version**-  v1.0.7 (based on Prusa Buddy 4.3.3-final)

**Custom firmware - Llama

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint**
  USB and Octoprint

**Describe the bug**
  I printed more than 20 test to compensate XY skew. And increase XY coefficient from 0.02 to 0.1. And diagonals length didn't change. I think this function isn't working.
A -  142.04
B - 140.65
I restore settings to default after install Llama firmware after original.
Tried to print via OctoPrint and USB.

SiegelDaniel, 2022-03-02:
Did you store settings to EEPROM ?

dagov42, 2022-03-02:
Yes, skew compensation works only in bed leveling mode, after that printer start printing - it doesnt work.

SiegelDaniel, 2022-03-04:
I've noticed the same problem with mine. I did the skew test print, logged in the values the spreadsheet gave me and afterwards the results remained the same. @matthewlloyd  Is this known yet?

matthewlloyd, 2022-03-04:
It's not known. I don't have a Prusa Mini any more so it's impossible for me to test and debug personally. I haven't made any changes to the skew correction code but it's possible Prusa changed something in the latest firmware which broke it when I rebased the changes. Certainly it was working on the previous version. I did take a quick look through the source code and see no reason why it would be broken. So this will need to be figured out by someone else. 

Some debugging ideas:

- Have you tried setting a high correction value with e.g. "M852 I0.20" then verifying that a square comes out as a parallelogram, either in a real print or manually with LCD-menu moves or "G1 X... Y..." commands?
- Are your skew settings showing up in M503?
- Do you have anything in your slicer's start G-code which could reset the settings?

SiegelDaniel, 2022-03-04:
Thank you for replying that quickly.

Regarding the slicer's start G-CODE:
`M862.3 P "[printer_model]" ; printer model check
G90 ; use absolute coordinates
M92 E415
M83 ; extruder relative mode
M104 S170 ; set extruder temp for bed leveling
M140 S[first_layer_bed_temperature] ; set bed temp
M109 R170 ; wait for bed leveling temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M204 T1250 ; set travel acceleration
G28 ; home all without mesh bed level
G29 ; mesh bed leveling 
M204 T[machine_max_acceleration_travel] ; restore travel acceleration
M104 S[first_layer_temperature] ; set extruder temp
G92 E0
G1 Y-2 X179 F2400
G1 Z3 F720
M109 S[first_layer_temperature] ; wait for extruder temp

; intro line
G1 X170 F1000
G1 Z0.2 F720
G1 X110 E8 F900
G1 X40 E10 F700
G92 E0

M221 S95 ; set flow`

Using pr0nterface, the skew settings do show up in M503. However, there is no reply from the printer on the M500 command.
[image]

I will try the high correction value now and report back in an hour.

dagov42, 2022-03-04:
Have you tried setting a high correction value with e.g. "M852 I0.20" then verifying that a square comes out as a parallelogram, either in a real print or manually with LCD-menu moves or "G1 X... Y..." commands?
 - Yes, I set 0.9 koefficient and probing failed cause probing mesh has distorsion and head goes away. I used lesser koeffcients around 0.2, mesh deformates but square diagonals doesn't change.
Are your skew settings showing up in M503?
- Yes, its the same settings as I set from display.
Do you have anything in your slicer's start G-code which could reset the settings?
 - No, I use start codes but it's not correct this settings.

SiegelDaniel, 2022-03-04:
I've verified with I=0.2 that the probe does indeed run off of the bed, which is why probing fails. I'm now trying with 0.1 again.
So it's safe to say that during probing / bed leveling the skew correction is certainly taking place.

matthewlloyd, 2022-03-04:
Ok, so the skew correction is clearly working in general, but might not be working when actually printing. Printing with a high correction factor to make the effect obvious, but not so high as to break mesh leveling, may be a good idea. Maybe the skew correction is being reset or turned off somehow by the Prusa firmware at the start of a print. Some more debugging ideas:

- Try printing (a) from SD card, and compare to (b) printing via OctoPrint. Skew correction might be working in one but not the other.
- Put an M503 in the G-Code you are printing, and check the output - i.e. after the print has already started.
- Put the M852 I command in the G-Code to see if that makes a difference.

SiegelDaniel, 2022-03-04:
Hi, I've printed the skew correction print with 0.1 for XY skew and it did not change a thing.
I'll try the other two hints tommorrow, that are great ideas. Thank you very much.

SiegelDaniel, 2022-03-24:
Hi, I've recently tried setting the skew correction again AFTER bed leveling in the printer's start GCODE. 
However, there seems to be some calculation error as a minor correction of 0,0015 on XY results in the print being placed at a whole different position on the bed. 

Do you have any ideas why that might be?

espr14, 2022-12-21:
Fixed in #29.

espr14, 2022-12-21:
> Hi, I've recently tried setting the skew correction again AFTER bed leveling in the printer's start GCODE. However, there seems to be some calculation error as a minor correction of 0,0015 on XY [...]

Skew center is in [0,0,0] not in the bed center. So if you print with Y = 100, there will be some X shift.

espr14, 2022-12-31:
Try this version https://github.com/matthewlloyd/Llama-Mini-Firmware/issues/19#issuecomment-1368281897.

espr14, 2023-01-11:
Or go directly to new project where the development will continue:
https://github.com/Llama-FW/Llama-P32-FW/releases/tag/v4.4.01

matthewlloyd, 2023-02-04:
Thanks for opening an issue. Unfortunately since I am no longer have time to maintain this project, I am closing all PRs and issues so that I can archive the repository.

closed 2023-02-04 by matthewlloyd