matthewlloyd/Llama-Mini-Firmware #21: [FEATURE REQUEST] Power Outage Recovery

Fra085, 2022-04-16:
Is it possible to implement the power loss recovery feature like the MK3 for example? Prusa promised this long ago but never happed for the MINI. It would be great to see this implemented in your firmware.

Thanks a lot.

SiegelDaniel, 2022-04-16:
@Fra085 To my understanding @matthewlloyd does not continue development for this firmware as he does not own a Prusa Mini anymore.

 I will try and port the llama firmware to the newest Marlin Version in June / July 2022 and will make the necessary changes for power loss too.

Fra085, 2022-04-16:
Oh ok, thank you. I will wait for your release then.

espr14, 2022-12-21:
Power loss is tricky for MINI because it lacks the HW device which is in MK3. There has to be different approach (voltage/current drop). This could appear in the original FW in the future.

matthewlloyd, 2023-02-04:
Thanks for opening an issue. Unfortunately since I am no longer have time to maintain this project, I am closing all PRs and issues so that I can archive the repository.

closed 2023-02-04 by matthewlloyd