matthewlloyd/Llama-Mini-Firmware #25: [FEATURE REQUEST] Change Load/Unload feed length

dagognatt, 2022-10-05:
Having changed the bowden tube when upgrading extruder, I ended up with a bit shorter bowden tube than the original. This leads to the loading-procedure ramming filament into the hotend at full speed before starting the purge process. 

It would be nice to see an option to adjust the length of the fast-feeding part of loading filament. (If this firmware is still active)

espr14, 2022-12-21:
Makes sense for direct drive mods as well.

Eldenroot, 2022-12-22:
Good idea

topchiyev, 2022-12-25:
Definitely a good idea!

matthewlloyd, 2023-02-04:
Thanks for opening an issue. Unfortunately since I am no longer have time to maintain this project, I am closing all PRs and issues so that I can archive the repository.

closed 2023-02-04 by matthewlloyd