matthewlloyd/Llama-Mini-Firmware #26: [ENHANCEMENT]update based on 4.4.0 fw

g8toparadise, 2022-12-11:
**Printer type** - [MINI]
Since 4.4.0 FW is out update your llama.
Unless you wait for a more stable release.

threepoint2, 2022-12-14:
i understand matthew lloyd sold his prusa mini and stopped developing this firmware; unless someone forks it and updates it, this is now dead.

espr14, 2022-12-21:
There are tons of changes. This will take some time.

Eldenroot, 2022-12-21:
Maybe better to create a PR for Official firmware and implement.some.features to base.

espr14, 2022-12-22:
4.4.0 is the biggest change for MINI (bigger BBF, WiFi etc.) since the beginning so it will need some proper testing.
PR #31

espr14, 2022-12-23:
Happy Christmas, here's gift for you.
@matthewlloyd PR #31 is ready for review. It compiles but I haven't tested it yet.
Any volunteer to test OctoPrint screen and connection?
4.4.0 removes 102,000 lines while adds 770,000 lines of code.

Eldenroot, 2022-12-23:
Great, thx

espr14, 2022-12-24:
Here's the [FW_MINI_Llama_alpha_based_on_4.4.0.zip](https://github.com/matthewlloyd/Llama-Mini-Firmware/files/10298302/FW_MINI_Llama_alpha_based_on_4.4.0.zip)

Eldenroot, 2022-12-24:
Thank you. What about other PRs  - are included or not?

espr14, 2022-12-24:
Other PRs are not in this.

prongato, 2022-12-25:
> Here's the [FW_MINI_Llama_alpha_based_on_4.4.0.zip](https://github.com/matthewlloyd/Llama-Mini-Firmware/files/10298302/FW_MINI_Llama_alpha_based_on_4.4.0.zip)
> Press knob during start to force loading. Bootloader >2.0.0 is needed. If you don't have it, install original FW 4.4.0 first.

I will help test it!  I have been so excited for this firmware because I am running a Mosquito and use the PID auto tune and like writing it to the EEPROM.  I will install it tomorrow.  What for format, etc do you want error reporting in?

prongato, 2022-12-25:
> Here's the [FW_MINI_Llama_alpha_based_on_4.4.0.zip](https://github.com/matthewlloyd/Llama-Mini-Firmware/files/10298302/FW_MINI_Llama_alpha_based_on_4.4.0.zip)
> Press knob during start to force loading. Bootloader >2.0.0 is needed. If you don't have it, install original FW 4.4.0 first.

BTW, if you plan on possibly taking over or forking Llama, just tell me where to donate!  I fell in love with this version of firmware because I like to tinker with my Mini and the default firmware is locked down.

do you know if they’ve fixed the mesh level bug on 4.4.0 yet?

prongato, 2022-12-28:
> Here's the [FW_MINI_Llama_alpha_based_on_4.4.0.zip](https://github.com/matthewlloyd/Llama-Mini-Firmware/files/10298302/FW_MINI_Llama_alpha_based_on_4.4.0.zip)
> Press knob during start to force loading. Bootloader >2.0.0 is needed. If you don't have it, install original FW 4.4.0 first.

- Tried to update the firmware on my Mini and get this error when rebooting [https://imgur.com/a/DkEHS01](url)

espr14, 2022-12-30:
There's a plan to create indepentent repository. I'll let you know then. I think the bug is not fixed and won't be fixed soon because of XL release.

I'll look at the error. But it seems it won't be easy so it won't be soon.

prongato, 2022-12-31:
> There's a plan to create indepentent repository. I'll let you know then. I think the bug is not fixed and won't be fixed soon because of XL release.
> 
> I'll look at the error. But it seems it won't be easy so it won't be soon.

Good to know.  I’m available and willing to help with any testing.  No hurry for me, I would prefer you port a latter release anyway because 4.4.0 has a mesh level bug I don’t think they’ve addressed yet.  

My biggest concern is a semi-consistent place to obtain unlocked firmware for my Mini so I have access to PID auto tunes, skew corrections, the ability to control the hotend fan percentage (I run a Slice Mosquito with a 25mm 5v fan on mine).   I appreciate any work you put into maintaining this firmware.  If you need donations or any support I can offer, please just let me know.

espr14, 2022-12-31:
Meanwhile, there's version based on 4.3.4 https://github.com/matthewlloyd/Llama-Mini-Firmware/issues/19#issuecomment-1368281897.

espr14, 2023-01-11:
Here's Llama's v4.4.01
https://github.com/Llama-FW/Llama-P32-FW/releases/tag/v4.4.01

espr14, 2023-01-19:
Here are the latest releases.
@prongato You can donate here: https://www.patreon.com/espr14

matthewlloyd, 2023-02-04:
Thanks for opening an issue. Unfortunately since I am no longer have time to maintain this project, I am closing all PRs and issues so that I can archive the repository.

closed 2023-02-04 by matthewlloyd